shooter64738/grbl-mega #1: Instructions For Use

bwanamukubwa, 2019-05-28:
Hi, i wasn't sure where to post this request but would it be possible to have a short video of the features you have contributed? 

I am interested in the backlash compensation and have a controller i'm willing to put a 2560 on and any user guide is much appreciated. 

Thank you in advance.

shooter64738, 2019-07-10:
Ide be glad to helpnif you still need it. I have been off line for health reasons but im getting better.

shooter64738, 2019-07-19:
To use this feature, set up the $140,141,142 values. Each value is the distance an axis needs to backlash comp. Compensation only happens when the axis direction changes. It will insert a rapid motion move before executing the programmed line. The distance it moves will be the $140,141,142 values for the axis that is moving. An axis that is not moving should be ignored.

JustZeus, 2019-10-30:
Hello, how can I use your program with an Arduino UNO R3?

yushiromarcel, 2022-02-24:
Thanks for this great work.
In the read me text, you speak about X, Y, Z and also A, B ($143, $144) for 4th and 5th axis. but I don't find $143 and $144.

Is there something wrong with that ?

Thanks !