shooter64738/grbl-mega #2: other axis moves when jogging axis changed

beautifulsmall, 2019-07-20:
using grbl-plotter. e.g. can jog x + and - 1mm ok but if then jog y, the x will move a (random?)multiple of 1mm ,ie 4mm then the y will  move its jog. the y will then jog ok until an x jog where the y will move before the x jog. 
only using on an xy system . 140,141 set to 0 or 0.01 same effect.
setup works ok with grbl-mega-edge, no hardware changes.   
arduino ide 1.8.4, grbl-plotter 1.2.4.9 , old grbl libraries deleated when swapping. no changes to config.h
settings
10,25,0,0,0,0,0
1,0.01,0.002,0
0,0,0,0,25,500,250,1.000
1000,0,0
4665,7350,250
75,75,50
25,25,10
200,200,200
0,0,0  (also tried 0.01,0.01,0)
great work, cheers, cheap ballscrew 0.18mm backlash on x, achme 60 year old on y with nut adjusted 0.005mm downham mini-jig borer conversion.

shooter64738, 2019-07-20:
Im sure its not your setup. It should work with an existing setup just fine. Ill set my mega back up and check it out.

shooter64738, 2019-07-20:
Get it set up and I do see the issue.. and I think I know where the bug is. :) Sorry!

shooter64738, 2019-07-20:
Update is checked in, give that a go. I am not seeing the behavior anymore. If you run into another issue let me know and I'll see if I can address it pretty quick.

beautifulsmall, 2019-07-21:
Yes thats fixed it. youve added a zero or even two to the price of my ballscrew. fantastic. Ironically its one small step for machine, one giant leap for machining. what a bizzare concidence. very happy. just cut a square within 0.005mm

beautifulsmall, 2019-07-21:
Sorry, getting over excited, i am out by the backlash. let me check again

beautifulsmall, 2019-07-21:
this issue is fixed but I'll open a new one for the adding backlash every axis swap not just direction swap.

shooter64738, 2019-07-21:
Ok I will close this issue, and look at the new issue. Thanks for testing this, My machine is down so all I can do at the moment is bench test it with servos. I'll keep on it till its working as expected!

closed 2019-07-21 by shooter64738