shooter64738/grbl-mega #3: backlash added for axis change not just direction change

beautifulsmall, 2019-07-21:
when changing axis the backlash is added. I have 140=0.2, jogging with grbl-plotter. when jogging x+,y+,x+,y+  it adds 0.2mm to every x jog even though no direction is changed.

shooter64738, 2019-07-21:
I've been testing this, but I've been unable to re-create it. I dont have a jogger, but I was alternating g0 moves. 
Can you try restarting your board, and trying this:
g0x1
g0y1
g0x2
g0y2
g0x0y0 (*Comp should happen here on both axis)
g0x1 (*Comp should happen here on X axis)
g0y1 (*Comp should happen here on Y axis)
g0x2
g0y2
If it behaves correctly it might be something related to it being a jog specifically although I dont know what that would be just yet. If you are seeing some other kind of behavior than listed above let me know. I have it reporting when a comp move needs to be executed, but Im not seeing it do any comp moves except when needed right now. I have my comp set to a crazy amount (50mm) so would def know if it adds one to an axis. 

I ported this over from another cnc control project I was working on, and it may take a little more tweaking to get it to work right. You can always turn it off temporarily by setting $140,141,142 = 0

beautifulsmall, 2019-07-22:
My example was perfectly wrong. My apologies.  I've re-tested and the additional backlash is only added to the negative step, when changing axis. so for x- ,y+or-, x-.
Also shows on the Y axis again only for -ve Y steps when the x axis has previously changed. 
the grbl-plotter outputs
$J=G91 x-1.000 F100  1st step, no added move
$J=G91 y-1.000 F100  y + or - 
$J=G91 x-1.000 F100  backlash added

ive also tried 
G0x4
G0y1
G0x3 added here ok
G0y2
G0x2  added here and shouldn't be. ect.
Ive got a DRO on the mill so its a great diagnostic help.

shooter64738, 2019-07-22:
Ahh.. let me do somw testing with that info. Should be pretty easy to produce and correct!

shooter64738, 2020-12-29:
Code and hex is updated. Erroneous print messages for BLC are gone, and correction to the BLC distance was made. Tested on a mega 2560. I'll close this issue now.

Thanks again!

closed 2020-12-29 by shooter64738