shooter64738/grbl-mega #4: GRBL Alarm:2 after 2. change of the direction

ChristianBrandtner, 2020-01-20:
in use:
the compiled hexfile Grbl_mega_back_compC.hex
$I
[VER:1.1g.20180813.Mega:]
[OPT:VNM+,35,255]

Steps:
1. homing the machine
2.**g0 x 100**
checking comp loop
ok
$G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
3.**g0 x 50**
checking comp loop
axis 0 set for compok
$G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
4.**g0 x 100**
checking comp loop
axis 0 set for compALARM:2
[MSG:Reset to continue]

the error is reproducible.

complete terminaloutput in attachement

[backlash_Error.txt](https://github.com/shooter64738/grbl-mega/files/4087668/backlash_Error.txt)

ChristianBrandtner, 2020-01-21:
the error is fixed.
normaly all targetvalues are negativ values, in case of backlash the targetvalue (backlash value) could be positiv. In this case "limits_soft_check" returns an error.
In file motioncontrol.c in function mc_line 

add
 		
if (!(pl_data->condition & (1 << PL_COND_FLAG_BACKLASH_COMP)))
			{limits_soft_check(target);}

that means, no limit check in case of backlash motion.

[motion_control.zip](https://github.com/shooter64738/grbl-mega/files/4093598/motion_control.zip)

closed 2020-01-21 by ChristianBrandtner

shooter64738, 2020-01-21:
Hey! I try to keep up with this but some permanent health issues are getting in my way. Glad you have it corrected. I will merge it into the tree