shooter64738/grbl-mega #5: wrong output in the copiled Grbl_mega_back_compC.hex

lookie25, 2020-03-22:
Hi. 
In the copiled Grbl_mega_back_compC.hex the line printString("checking comp loop\r"); in backlash.c is not comment out. So every gcode command return the mesage "checking comp loopok". This make problems with gbrl controler software.
Perhaps you can update the compiled hex.
You make a great job, your grbl version works perfect for me!
Best reguards
Thomas

ChristianBrandtner, 2020-03-22:
[grblBacklash012020.zip](https://github.com/shooter64738/grbl-mega/files/4365919/grblBacklash012020.zip)

Try this Version, it's running, see also closed Issue #4

Regards Christian
[grblBacklash012020.zip](https://github.com/shooter64738/grbl-mega/files/4365935/grblBacklash012020.zip)

shooter64738, 2020-04-30:
Woops! I'll remove that debug message!

TomMandle, 2020-11-20:
> Woops! I'll remove that debug message!

Hi Shooter.  Thank you for adding the backlash comp.  I installed the grblBacklash012020.zip but it is very slow.  I changed the backlash numbers to 0 and it did not help.  I installed the hex using loader.  The mega board is fine because I ran another version  and it was very fast.  

Thank you again for all of your work.

shooter64738, 2020-12-29:
Code and hex is updated. Erroneous print messages for BLC are gone, and correction to the BLC distance was made. Tested on a mega 2560. I'll close this issue now.

Thanks again!

closed 2020-12-29 by shooter64738