shooter64738/grbl-mega #7: Is it stable version of grbl mega2560

arlimbad222, 2020-07-28:
Can anyone tell is this GRBL is stable currently I use grbl mega 5x . Now my machine screw has a small amount of backlash and need a backlash comp.

shooter64738, 2020-07-29:
This version of grbl was forked from the main grbl branch so should be stable. The only modifications I made was to port the BL comp code from my control to this one. It shoukd be as stable as the original version.

arlimbad222, 2020-07-29:
First thankyou so much for quick response .

This is aman from india and I want to use your code to control my cnc .
Have you taste backlash comp  Completely is this any error or modification
needed . Currently I am using android app as a g-code sender.

And do you have any instructions how to use backlash comp etc....

closed 2020-07-31 by arlimbad222