shooter64738/grbl-mega #8: Soft limit triggered

arlimbad222, 2020-07-31:
Hello sir I am testing my machine with your code & I find when I put some value on $140-$141 and try to jog my machine in + or - direction . But I can't move axis on both directions when I hit X+ then it's move but when I hit X- it gives a soft limit alarm & my machine have only one limit switch so I can't turn off soft limits.

Please please help me I almost give my 3 days to configure your code & grbl settings.

arlimbad222, 2020-07-31:
>>> $J=G21G91Y-1F1000
ok
>>> $J=G21G91Y1F1000
[Error] An error was detected while sending '$J=G21G91Y1F1000': (ALARM:2) Soft limit alarm. G-code motion target exceeds machine travel. Machine position retained. Alarm may be safely unlocked. Streaming has been paused.
[MSG:Reset to continue]

This happens when axis change direction please help

shooter64738, 2020-07-31:
The configuration is identical to standars grbl aside from being able to specify the backlash distance. Backlash motions shoukd not avtually update position on the system because itbhasnt really moved.

What are the values you have set fot Y axis backlash comp?

arlimbad222, 2020-07-31:
Oh sorry my problem was solved I learnt from one of your closed issue I think someone make an motion_control.c I just replace motion_control.c and everything will be fine and working for me now and again thankyou so much for making backlash comp it helps me a lot .

ChristianBrandtner, 2020-07-31:
only the sourcecode in this repository has to be updatet with my fix
https://github.com/shooter64738/grbl-mega/issues/4#issuecomment-576894974

arlimbad222, 2020-07-31:
Thankyou so much for your fix . I am not a computer guy so basically I am mechanical engineer and I can't even write a one line of Arduino code . So you guys are like God for me and if anyone want to know about my machine mail me I will share some photos & also my friend edit one android app for me to control my machine with grbl .

I made a new startup in india . & I use grbl as a my controller I also make my own pcb to control my machine.

shooter64738, 2020-12-29:
Code and hex is updated. Erroneous print messages for BLC are gone, and correction to the BLC distance was made. Tested on a mega 2560. I'll close this issue now.

Thanks again!

closed 2020-12-29 by shooter64738