shooter64738/grbl-mega #9: mess programs

arlimbad222, 2020-08-01:
hello sir i test my machine today with dial indicator it runs awesome in test run or joging mode . but when i start my first program it behave very strange my whole design running in taper on X axis i dont know why but when i remove backlash comp . it work perfect . any suggestion.

romlel, 2020-12-28:
Hello. I have the same problem here. For "simple moves", I don't see any problem, but with a program CAMed with Fusion360 , with some spiral moves and so on (ramping and adaptative clearing), an offset qucikly builds up in the machine VS the program on screen (using bCNC).  Perhaps spirals are not recognized properly ? I tried with full linearization of arcs in case G2/G3 would be a source of pb, but no better result it seems. I'll have a look at the code and try to diagnose, but for now, didn't understood yet of all of this work :) GRBL is not a tiny project...

@shooter64738 , if you had some theories, even "top of your head", about why this could of behavior could appear - that would allow us to try to "avoid the situation" (at least), or even find a potential fix.

thanks !

shooter64738, 2020-12-29:
I'll get the comp changes uploaded, and this should be corrected.

shooter64738, 2020-12-29:
Code and hex is updated. Erroneous print messages for BLC are gone, and correction to the BLC distance was made. Tested on a mega 2560. I'll close this issue now.

Thanks again!

closed 2020-12-29 by shooter64738