shooter64738/grbl-mega #10: Compiled version does not suppress print statements

colinmsnow, 2020-10-15:
Hi,

First off, thank you so much. This is an amazing improvement to GRBL and it was incredibly helpful with my build.

I noticed that the current compiled binary appears to be a bit off from the rest of the repo in that it does not suppress print statements related to checking the backlash values. I'm not sure how applicable this problem is, but those statements cause Universal Gcode Sender to crash for me when sending programs. I presume this is just because of the giant overflow in information clogging up the serial line.

I compiled the project from source and it works great, but the binary should probably be updated for those who don't know how to do that.

If you want I can pull request the new binary.

Thanks,
Colin

ChristianBrandtner, 2020-10-15:
before you pull the new binary, have a look to this issue: 
https://github.com/shooter64738/grbl-mega/issues/4#issuecomment-576872430
it isnt corrected in source code here.

Christian

shooter64738, 2020-12-29:
I have a correction for this I just didnt get it pushed up. I'll work on getting that up here. Its been a crazy busy year for me.

shooter64738, 2020-12-29:
Code and hex is updated. Erroneous print messages for BLC are gone, and correction to the BLC distance was made. Tested on a mega 2560. I'll close this issue now.

Thanks again!

closed 2020-12-29 by shooter64738