shooter64738/grbl-mega #11: k

hktt-git, 2020-12-28:


romlel, 2020-12-28:
Hi. Seems that we are having a similar problem. I just asked some questions on another issue.

I'm not sure I understood your proposition. Could you please copy/paste a snippet of code that should be ok according to you ? I'll have a look on my side also.

Thanks !

hktt-git, 2020-12-28:
Thank you for your reply.
I suggest fixing "planner.c:393"

from
```
if (target_steps[idx]!=0)
    back_lash_compensation.last_comp_direction[idx] = 1;
```
to
```
if(delta_mm != 0.0)
    back_lash_compensation.last_comp_direction[idx] = 1;
```

romlel, 2020-12-28:
Ah, yes :) That what one of the fixes I tried from your first comment, and firsts tests seems good.
I added on my working copy a "debug variable" to count the number of compensations done in each direction over time, and with your fix, they stay "in equilibrium", before, the two counters were drifting more and more apart (so compensations were not symetrical.

i'll take it to the physical CNC tomorrow on a real use case, and keep you - and the main author or course- posted
Thanks !

romlel, 2020-12-29:
Well, it seems to work well. I have to make more tests but first one was positive
Thanks a lot !

@shooter64738 : Do you think this fix is correct ? If so, could you please integrate it in the main code for everyone to enjoy ? thanks

shooter64738, 2020-12-29:
Sure guys! I have a few tweaks I need to push up to this repo. I'll get this change added as well and compile a new version. Thanks for the efforts.  Also if you have an stm32h7 board the talos project is a much bigger cnc control. I never could get enough interest in it for anyone to lend a hand, but its a commercial grade cnc control. Spent 6 years on it, and Im just running out of interest in it.

shooter64738, 2020-12-29:
Code and hex is updated. Erroneous print messages for BLC are gone, and correction to the BLC distance was made. Tested on a mega 2560. I'll close this issue now.

Thanks again!

closed 2020-12-29 by shooter64738