shooter64738/grbl-mega #12: Pinout of the grbl-mega

CognitiveConnexionist, 2021-10-30:
Hello,
thanks for putting out a version of grbl that has backlash correction. I have not been able to test it because I could not find the pinout or connections that go to/from the mega board.
Can you please confirm that backlash works for arc corrections without transforming arcs into short lines?
Thanks for your answers!

shooter64738, 2021-10-31:
The pinout wasnt changed and should be the same as the standard grbl kega version.

Grbl converts arcs i to short line segments as its way of interpolation of arcs. This version functions the same way and as each segment is queued, if a segment will cause a direction change in the axis, backlash comp will run.

shooter64738, 2021-10-31:
The pin names are in the comments on the right.

#define STEP_DDR      DDRA
  #define STEP_PORT     PORTA
  #define STEP_PIN      PINA
  #define X_STEP_BIT    2 // MEGA2560 Digital Pin 24
  #define Y_STEP_BIT    3 // MEGA2560 Digital Pin 25
  #define Z_STEP_BIT    4 // MEGA2560 Digital Pin 26
  #define STEP_MASK ((1<<X_STEP_BIT)|(1<<Y_STEP_BIT)|(1<<Z_STEP_BIT)) // All step bits

  // Define step direction output pins. NOTE: All direction pins must be on the same port.
  #define DIRECTION_DDR     DDRC
  #define DIRECTION_PORT    PORTC
  #define DIRECTION_PIN     PINC
  #define X_DIRECTION_BIT   7 // MEGA2560 Digital Pin 30
  #define Y_DIRECTION_BIT   6 // MEGA2560 Digital Pin 31
  #define Z_DIRECTION_BIT   5 // MEGA2560 Digital Pin 32
  #define DIRECTION_MASK ((1<<X_DIRECTION_BIT)|(1<<Y_DIRECTION_BIT)|(1<<Z_DIRECTION_BIT)) // All direction bits

  // Define stepper driver enable/disable output pin.
  #define STEPPERS_DISABLE_DDR   DDRB
  #define STEPPERS_DISABLE_PORT  PORTB
  #define STEPPERS_DISABLE_BIT   7 // MEGA2560 Digital Pin 13
  #define STEPPERS_DISABLE_MASK (1<<STEPPERS_DISABLE_BIT)

  // Define homing/hard limit switch input pins and limit interrupt vectors. 
  // NOTE: All limit bit pins must be on the same port
  #define LIMIT_DDR       DDRB
  #define LIMIT_PORT      PORTB
  #define LIMIT_PIN       PINB
  #define X_LIMIT_BIT     4 // MEGA2560 Digital Pin 10
  #define Y_LIMIT_BIT     5 // MEGA2560 Digital Pin 11
  #define Z_LIMIT_BIT     6 // MEGA2560 Digital Pin 12
  #define LIMIT_INT       PCIE0  // Pin change interrupt enable pin
  #define LIMIT_INT_vect  PCINT0_vect 
  #define LIMIT_PCMSK     PCMSK0 // Pin change interrupt register
  #define LIMIT_MASK ((1<<X_LIMIT_BIT)|(1<<Y_LIMIT_BIT)|(1<<Z_LIMIT_BIT)) // All limit bits

CognitiveConnexionist, 2021-11-01:
Thank you very much for your answer, just another question, while one axis is getting its backlash corrected, does the other axis stop moving until the backlash has been corrected?

shooter64738, 2021-11-01:
Yes. Because of the way motuons are queued in grbl, the backlash motion will only move the axes that change direction. The comp should be fast taking only a millisecond ( unless you are comping extreme backlash). The tool should not have a noticeable 'rest' during compensation. 

My other controller (Talos) behaves differently on backlash comp, but this was the only practical way to fit it into base grbl.

skitter155, 2021-12-03:
Is it possible to use a CNC shield for the uno on the mega with this code? Or are the pins incompatible?

shooter64738, 2021-12-03:
As far as I am aware the pin assignments between the Uno and Mega version are incompatible.

TomMandle, 2021-12-03:
Hi,
I used the GRBLDuino Mega Lite-GRBL 1.1 Arduino CNC Shield from Tindie.  It worked well with my Bridgeport CNC  conversion.
The backlash version is fantastic.  Thank you for creating it.
A request for fixing a small defect.  When the table comes up to enter the parameters, the x,y,z backlash items are missing the "description" for $140, $141, $142.
Thank you again.

shooter64738, 2021-12-03:
> Hi, I used the GRBLDuino Mega Lite-GRBL 1.1 Arduino CNC Shield from Tindie.  It worked well with my Bridgeport CNC  conversion. The backlash version is fantastic.  Thank you for creating it. A [...]

If I get some spare time, I'll add descriptions. 

I spend most my time working on my other control, and when Ive got a feature in it I can port to grbl, I try to get it in there.