shooter64738/grbl-mega #17: 3 axis complied, 4th and 5th Please.

Earsmith, 2022-04-10:
I used xLoader to uploaded the 3 axis hex. Then connected GrblGRU and it responded as expected. But I have a Taig 4th axis machine. After spending more time than I would like to admit troubleshooting Arduino IDE just to compile for the 4th axis I see this is the only sane thing to do. Would you, please, compile for the 4th and the 5th. Thanks in  advance.

shooter64738, 2022-04-10:
You need the backlash compensation enabled for the 4th-6th axes? Just checking, Im not 100% sure which feature you were asking about.

Earsmith, 2022-04-11:
Yes on the backlash enabled. 
When I ran across your fork it was a pleasant surprise that with what I already own I could update my Taig to the latest features in CNC.  Need to purchase the TMC2160-OC stepper drivers.  I've used my mill very little through the years as I crashed it more than once, lack of limit switches, novice experience and somewhat confusing nature of Mach3. A CNC mill is not in the 'toy' category. On the other hand 3D printers have limit switches and never crashed my Makergear M2, that thing is a beast and over eight years old controlled with a MKS GEN L and TMC2208's in UART mode. It's so quiet the fan is the loudest part of the printer.  That's why the choice of the TMC2160-OC and if I want to update a more powerful controller easy swap. The Taig uses a 20 TPI lead screw so high feed rate may not be possible with the limitation of the Atmel 2560 16MHZ clock speed running up to 5 axis simultaneity. If it seems I'm being demanding trying not to. Thought I would give as much info about how I intend to use your MEGA 5X version. Because as you know most CNC hobby is belt driven with almost no backlash and CNC lead screws needs backlash compensation. Again, thanks in advance.

shooter64738, 2022-04-11:
No problem. I'm moving to a new PC so I need to rebuild my avr setup. I work a lot in the stm32 world right now. I've got a 32bit arm control I'm slowly getting g finished (not grbl though). It's 6 axis and I'm really close to breaking 1mhz step rates if your drivers can do that.

I'll get my avr environment set back up and get the 4,5,6 axis comp working and publish it.

Earsmith, 2022-04-14:
Since your not doing much in the avr world it really begs the question. Should I just bite the bullet and go with something in the range where you are moving, hmm. How much is too much and is there unlocked potential in my mill with a more powerful controller. The Taig is precise but trades off somewhat in speed. I've almost driven myself nuts asking these questions. So if I can use what I already own, nothing lost. Take your time and since you have a mill any advice would be appreciated. Oh, I kicked myself after not including bed probing in the wish list. Your efforts are appreciated.

Earsmith, 2022-04-16:
Warning: Pest Mode.
I don't know where you expertise lays so I'm just going to just throw this out there. 
Are you familiar with the 3D printer firmware Klipper. it has something called

Resonance Compensation:
https://www.klipper3d.org/Resonance_Compensation.html
using a ADXL345 3-Axis Digital Acceleration Module for Arduino 

The application to a CNC mill would measure the steps/ pulses between the lead screw turning and bed movement. Pretty simple statement but I'm sure the math could be daunting. The application could be as simple as giving a readout and the user could manually enter the values.  But also create a log during milling as it warms up and tolerances change.  Could even warm up your mill with start up gcode and find the most stable state.  I could go on and on. Sounds like, hoping you could pull something like this off.

shooter64738, 2022-04-16:
My expertise like somewhere between experienced, and expert on some portions. For example, I know all thr required parameters for certain gcode motions, and then there are some concepts I'm just scratching the surface one.

The new control system (Talos) is built and configured like a commercial grade control. (It's all based on fanuc control methods). The control has pretty much all the features you would find on a commercial machine.

The biggest difference is that the Talos control